qitechgmbh/control #670: Cycle time is speeding up

1cedsoda, 2025-09-04:
As James pointed out, our cycle time is ever speeding up https://github.com/ethercrab-rs/ethercrab/issues/293#issuecomment-3189301400 in a strange pattern, this is likely because the `LoopThrottle` algo is wrong. We can fix `LoopThrottle` or just sleep 1ms.

<details>
<summary>LLM Opinion</summary>
I can see the issue with the `LoopThrottle` implementation. Looking at the code, there's a fundamental problem with the current adaptive logic:

`control/control-core/src/helpers/loop_trottle.rs#L45-50`
```
// Calculate error in nanoseconds (signed)
let error_nanos = target_cycle_time_nanos as i128 - avg_actual_cycle_time_nanos as i128;

// Update current delay, allowing zero or negative values to become zero
let new_delay_nanos = (self.delay.as_nanos() as i128 + error_nanos).max(0) as u64;
```

The problem is that this is a **purely additive correction system** without proper bounds or recovery mechanisms. Here's what happens:

1. **When the loop body takes longer than expected:**
   - `error_nanos` becomes negative (target < actual)
   - The delay gets reduced each cycle: `new_delay = current_delay + negative_error`
   - This continues until `delay` reaches 0

2. **Once delay reaches 0:**
   - The system can no longer compensate for slow loop bodies
   - Even if the loop body becomes faster later, the delay stays at 0
   - There's no mechanism to increase the delay back up when conditions improve

3. **The cycle time "gets shorter and shorter"** because:
   - The delay keeps decreasing towards 0
   - But the actual work time remains constant
   - So the total cycle time approaches just the work time (minimum possible)

**The core issue:** The current implementation can only decrease delays, never increase them back to maintain proper timing.

Here's a better approach that would fix this:
`control/control-core/src/helpers/loop_trottle.rs#L45-50`
```rust
// Calculate error in nanoseconds (signed)
let error_nanos = target_cycle_time_nanos as i128 - avg_actual_cycle_time_nanos as i128;

// Use proportional correction instead of additive
// This allows the delay to both increase and decrease as needed
let correction_factor = 0.1; // Adjust this for responsiveness vs stability
let correction = (error_nanos as f64 * correction_factor) as i128;
let new_delay_nanos = (self.delay.as_nanos() as i128 + correction).max(0) as u64;
```

Or even better, reset the delay based on the error:
`control/control-core/src/helpers/loop_trottle.rs#L45-50`
```rust
// Calculate what the delay should be to hit target timing
let work_time_nanos = avg_actual_cycle_time_nanos as i128 - self.delay.as_nanos() as i128;
let required_delay_nanos = (target_cycle_time_nanos as i128 - work_time_nanos).max(0) as u64;
let new_delay_nanos = required_delay_nanos;
```
</details>

andergisomon, 2025-09-04:
Another potential lead:

https://github.com/qitechgmbh/control/issues/500#issuecomment-3245892341

kraemr, 2025-09-07:
https://github.com/qitechgmbh/control/issues/676

andergisomon, 2025-09-07:
Finally😅

kraemr, 2025-09-08:
https://docs.rs/ethercrab/latest/ethercrab/struct.Timeouts.html : wait_loop_delay: Duration

Polling duration of wait loops.

Some operations require repeatedly reading something from a SubDevice until a value changes. This duration specifies the wait time between polls.

This defaults to a timeout of 0 to keep latency to a minimum.

kraemr, 2025-09-08:
When both of these are merged ill test the cycle times again. I have seen that Loopthrottle is most likely to blame.

closed 2025-09-08 by kraemr